q3k/chubby75 #3: Add links on where to get boards?

mithro, 2018-12-17:
Where can you get these boards from?

enjoy-digital, 2018-12-18:
I got mines from Amazon: https://www.amazon.com/Linsn-Receive-Card-RV901T-Controller/dp/B012JLZG10/

jeanthom, 2018-12-20:
I got mine from Aliexpress: https://www.aliexpress.com/item/New-Arrival-LINSN-RV901-Full-Color-RGB-LED-Display-Screen-Receiving-Card-12-HUB75-included-Crazy/738805774.html?spm=a2g0s.9042311.0.0.29364c4dy5kava

It took me 21 days to get it delivered to France.

danielkucera, 2018-12-29:
https://www.aliexpress.com/item/LINSN-RV901-Receiving-card-suitable-for-all-kind-of-HUB-board-work-with-TS802D-Sending-card/32899129837.html

zoobab, 2019-01-05:
There also some on Taobao:

https://item.taobao.com/item.htm?spm=a21wu.10013406.0.0.331e2ee4kTzWb0&id=548479172206