q3k/chubby75 #8: LEDG & button ?

jeanthom, 2018-12-23:
I traced the schematic of the button next to JP2 and S1 :

[image]

That schematic feels weird, maybe I've made a huge mistake ? What do you think ?

Also, I noticed that P4 from the FPGA isn't connected to the button but rather to U600:B7.

q3k, 2019-01-26:
Here's my tracing of this region:

![image](
https://q3k.org/u/fe5c9d2b3bf746c0891d94e858186cb648acca66ed37a9848581f86389baa494.png)

q3k, 2019-01-26:
The marking on Q503 is `1AM`, which points to a [MMBT3904LT1 NPN transistor by ONSemi](https://www.onsemi.com/pub/Collateral/MMBT3904LT1-D.PDF). Labeling BCE on the trace:

[image]

q3k, 2019-01-26:
I've stared at this for a while, and it looks exactly like the schematic @jeanthom posted.

So yeah, that's a weird circuit. I assume Q503 is used as a diode here, maybe also to limit C->E current drained by the button or when JP2.2 is tied low? I assume there's some factory use of JP2 (it's one of the few populated connectors) where this whole mess make sense.

I don't think we can actually read the button state via F7...

q3k, 2019-01-26:
Correction: we probably can read the button if we just set F7 as a high-impedance input - pressing the button then will cause F7 to go to 0Vish via Q503. When the button is not pressed, F7 is at around 2.8V.

q3k, 2019-01-26:
Documented in https://github.com/q3k/chubby75/commit/9f0e47c58e9b4f4853c8e42d5072594c361b15b8

closed 2019-01-26 by q3k