q3k/chubby75 #10: BGA pinout

danielkucera, 2018-12-29:
I'm little bit pissed of because I have been reversing this board since summer and now I've discovered this repo. :D But never mind, here's a doc I've been able to produce:

https://docs.google.com/spreadsheets/d/1L9uFN9jnsFd8CrgWXQyUStC2SsfoM5j2a6bWrxqi1Ko/edit#gid=0

Just this week I started to grind off and document PCB layers of this board. I have first layer off, slowly reaching the second one. Are there any more layers which need to be discovered or can I throw the board to trash?

NiklasFauth, 2018-12-29:
Nope, there are only four layers, from top to bottom: Signal, GND, Signal, Power.
So it is only interesting to sand off the bottom copper plane and this has already been done.

danielkucera, 2018-12-29:
Okay, thanks, at least I've found out soon enough :)

closed 2018-12-29 by danielkucera