q3k/chubby75 #16: datasheet for the Broadcom B50612D

Mar10us, 2019-05-24:
Any idea where i can find the datasheet for the ethernet phy **Broadcom B50612D** used on this board?

danielkucera, 2019-05-24:
I know there is one for  B50610 but I was not able to download it anywhere. http://www.51hei.com/bbs/dpj-122718-1.html

smunaut, 2020-01-17:
https://www.eevblog.com/forum/projects/where-can-i-get-the-b50610c1kmlg-datasheet/?action=dlattach;attach=902234

iDoka, 2021-12-03:
B50612D at least exists on chinese resources like https://download.csdn.net/download/mydream_sky/11088168
But I can't to download it.

derekmulcahy, 2021-12-04:
The B50612D data sheet is available at [https://github.com/wuxx/Colorlight-FPGA-Projects/doc](https://github.com/wuxx/Colorlight-FPGA-Projects/blob/master/doc/B50612D-datasheet.pdf)

iDoka, 2021-12-05:
@derekmulcahy indeed!
Thanks for sharing and it seems google not so good indexing github