q3k/chubby75 #17: EEPROM dump

Dnstje, 2019-08-24:
Hello, 
Do you have the EEPROM dump from this board and likely to share? It seems I bricked the one on my board.

Thanks,
Dnstje.

danielkucera, 2019-08-30:
[orig.zip](https://github.com/q3k/chubby75/files/3561684/orig.zip)

sjoerdtimmer, 2021-01-16:
Quick question: how did you extract this? I just received a v7.0 board and the first thing I wanted to do was to save a copy of the current firmware. I used Diamond with the "SPI flash background programming" option and the "SPI Flash Read and Save" operation but the IDCODE is wrong. 
Here is a screenshot of the settings:
[image]

And this is the Output:
```
Starting: "pgr_program run"

INFO - Check configuration setup: Start.

INFO - JTAG Chain Verification. No Errors.

INFO - Check configuration setup: Successful.

INFO - Device1 LFE5U-25F: W25Q32JV: SPI Flash Read and Save

Initializing...

IDCode Checking...

ERROR - Verification Error...when Processing function: 'CHECK_ID'

ERROR - Operation: unsuccessful.

ERROR: pgr_program failed.

ERROR - Programming failed.
```

sjoerdtimmer, 2021-01-16:
Self-reply: I figured it out. The ecp5 releases the flash interface as soon as the bitstream is loaded. To retain flash access after boot you must set SPI_MASTER to ENABLED in the global preferences. In case anyone is running into the same issue:
[image]

I feel a bit ashamed because I knew the ecp5 did that but I thought the default was the other way around:s

sjoerdtimmer, 2021-01-16:
In addition, my board has a different firmware image.
Maybe they changed it?

[backup.zip](https://github.com/q3k/chubby75/files/5824981/backup.zip)