q3k/chubby75 #20: Support for the RV908

Cellgalvano, 2019-10-10:
Hi!

I'm currently looking into buying some of these receive cards and I've seen, that there is a presumably newer Board called RV908. It looks like the RV901 with the HUB75 daughter board integrated. 
Would it be possible to add support for this version too?

Regards!

saeugetier, 2019-10-10:
The RV908 contains the same FPGA. If you attach the HUB75 daugther card to the RV901, you basically will get the same device. Signals are the same. So the Ethernet Phy and so on are connected to the same FPGA pins. I assume that RV901 and RV908 are using the same binary configuration file, when used as a LED driver board.

I have tested the RV908 and implemented my own HUB75 driver. You can use the signals from the hardware description of the RV901.

tomverbeure, 2019-10-20:
I bought the RV901 and the RV908, but didn't have any luck finding the JTAG connections. 

Did just reprogram the SPI flash with a flash programmer or did you map out the JTAG pins?

Tom

saeugetier, 2019-10-20:
You have to remove some solder stop on top of vias. I took a photo from my RV908. You can access the signals on the back side.

I could to a pull request, if the information of the RV908 was requested.

[image]

tomverbeure, 2019-10-20:
The PCB of my RV980 is significantly different. For example, the LEDG/3.3V/LEDR/KEYH/KEYL connector is much more to the left. Etc.

For me, it's not so much a matter of removing the solder mark from the vias, as figuring out which on. 

On the RV901T, the JTAG test points are much larger than the vias. On my RV908, there are no test points to be found, only hundreds of vias.

It's not clear on your picture above if you had test point for the JTAG signals or only vias?

Tom

saeugetier, 2019-10-21:
It seems that there are two versions of the RV908: RV908T and RV908M32.

I bought this board directly in Shenzhen in March this year and modified it right after return. So I can't remember quite well, what I habe done. I think mine had no test points. But the vias were on the same spot as on the RV901. So I guessed that it is the same signal. And I had luck. 

The back side of my RV908M32 looks very similar to the RV901. Maybe the design is derived from the RV901.

Cellgalvano, 2019-10-21:
I just bought a RV908M32, until is is delivered I'm looking for the differences of the different board versions.

https://controller-led.com/products/linsn-led-receiving-card-rv908-rv908h32-rv908m32-led-display-controller?variant=25457148821604
This site shows two version next to each other with a different PCB layout.

oramasdj, 2020-03-16:
> Hi!
> 
> I'm currently looking into buying some of these receive cards and I've seen, that there is a presumably newer Board called RV908. It looks like the RV901 with the HUB75 daughter board integrated.
> Would it be possible to add support for this version too?
> 
> Regards!

Hi, I connected one rv908m32 by a lan cable to my laptop, I configured netcard in the ledstudio program the 6 p4 panels start flashing, and then I went to the sender configuration, after send the rcg  file only by pressing sending to receiver button the panels got worst, so  i move something at sender option and I pres the send to sender button and the panels turned off do I got worry and I turned off the dc voltage to the board and panels, then I turned on again and I pressed the test button on the sender and nothing happened on the panels there is no test function and when I open the ledstudio there is no green led blinking on the receiver board, So I got a ts802d and tried to send the bin file to the receiver card trought the sending card but the board it only appeared with xx without the model the receiver board has no lan data transmition at all, even if I disvonnect the lan cable and reconnect it, the lan status on the board is off or there is no green led blinking, only on the laptop the lan status is active. How I could reprogram the board or repair by rs232 conection?

tomverbeure, 2020-03-19:
I've added a link to this issue in the main README so that it's more discoverable by those who might be googling for it, but I don't expect any more work on the RV908 board since all work has moved to the more interesting ColorLight 5A-75B board.

Closing this issue.

closed 2020-03-19 by tomverbeure

oramasdj, 2020-03-21:
I cant fi g the readme

> I've added a link to this issue in the main README so that it's more discoverable by those who might be googling for it, but I don't expect any more work on the RV908 board since all work has moved to the more interesting ColorLight 5A-75B board.
> 
> Closing this issue.

So, are there anyway or option to reflash or recover the rv908m32 receiver card?

oramasdj, 2020-03-21:
> I cant fi g the readme
>
>
> So, are there anyway or option to reflash or recover the rv908m32 receiver card?

Bytheway the 5a-75b only supports 256x256 pixels vs rv908m32 1024x256

richardklingler, 2021-11-23:
Just got me a RV908M32 board as well...

Trying to detect it with urjtag just gives me:

`IR length: 8
Chain length: 1
Device Id: not supported (bit 0 was not a 1)
jtag> print
 No. Manufacturer              Part                 Stepping Instruction          Register                        
-------------------------------------------------------------------------------------------------------------------
   0                                                         (none)               (none)                          
`

Any trick to make this board communicate via JTAG correctly?

tyalie, 2024-09-21:
For anybody who also ends up here: The current revision of the RV908M32 does not use a Xilinx FPGA anymore. They have switched to an Anlogic FPGA. In my specific case it was an Anlogic EG4S20BG256. What distinguishes these from the Xilinx based boards is the missing external SDRAM. But as it is still only a revision other things like the SPI flash programming header work as they did before.

davemaster, 2024-09-22:
> For anybody who also ends up here: The current revision of the RV908M32 does not use a Xilinx FPGA anymore. They have switched to an Anlogic FPGA. In my specific case it was an Anlogic EG4S20BG256. [...]

Greetings,

Have You got a picture, video, link.... ? Can use any other PGA chip, but, it will do the same as the older RV908 boards, right?

tyalie, 2024-10-02:
> Have You got a picture, video, link.... ? 

Can you clarify of what?

> Can use any other FPGA chip, but, it will do the same as the older RV908 boards, right?

I think specific to the use case that LinsnLED envisioned: Yes. But I think the specs are ever so slightly different.