q3k/chubby75 #21: better way to fix the JTAG!?

samsoniuk, 2019-10-23:
Hi colleagues,

I tested the mounting of the JTAG connector glued in the edge of the board, but I found that this configuration is not so "solid" from the mechanical point of view. Also, the JTAG connector in line requires use the color JTAG cable, which it needs to be wired pin by pin in the correct order. 

After some quick tests, I found that is possible use the 10-pin JP3 connector, in a way that the standard 14-pin flat cable found in the ultra-low-cost "xinglinx" JTAG adapter can be directly used:

[image]

Of course, although the pin numbers are labelled wrong (JTAG pin 1 connected to PCB pin 9, JTAG pin 2 connected to PCB pin 2, etc until all ten first pins are connected), the pin order from JP3 in the PCB is exactly the same as the first 10 pins from the 14-pin JTAG connector, which make it very quick and practical to connect:

VREF(9) TMS(7) TCK(5) TDO(3) TDI(1)
GND(10) NC(8) NC(6) **5V(4)** NC(2)

VREF(1) TMS(3) TCK(5) TDO(7) TDI(9) **NC(11) HALT(13)**
GND(2) GND(4) GND(6) GND(8) GND(10) **GND(12) GND(14)**

The NC pins are not connected in the PCB and can be connected to the JTAG GND, but the **5V(4)** is powered by the +5V in the PCB, which means that is better bender or cut this pin in a way that is not physically connected to the flat cable, in order to avoid a short circuit in the JTAG adapter between the GND and 5V!

[image]

Well, in the other side of the board, the main advantages are that the JP3 is better fixed via a THT connector, the GND is already present i the pin 10 and the 3V3 can be routed from a better location:

[image]

Best regards,
Marcelo

tomverbeure, 2020-01-09:
I think this is great!

Do you think you can convert these instructions into a separate documentation file and create a pull request?

Tom

samsoniuk, 2020-07-28:
Here the proposed changes and updates:

https://github.com/darklife/chubby75/blob/master/rv901t/README.md
https://github.com/darklife/chubby75/blob/master/rv901t/doc/getting_started/improved_jtag_getting_started.md

tomverbeure, 2020-07-28:
Changes applied. Closing.

closed 2020-07-28 by tomverbeure