q3k/chubby75 #22: Reverse-engineer the LED matrix driving Logic in FPGA

Hazeline2018, 2019-12-03:
Hi, 

This is a great work, guys! But it'll be even greater if the original LED matrix driving logic in FPGA can be reverse-engineered and shared here! I'm currently working on a project to develop a FPGA LED matrix display driver using Xilinx LX9. Would love to have some reference codes handy. 

Thanks!

MarcelWaldvogel, 2019-12-30:
Let me try to answer your question in parts, going from output (LED matrix) to input (GbE):

1. The output format is HUB75, which is e.g. [documented and implemented by Henner Zeller](https://github.com/hzeller/rpi-rgb-led-matrix).
1. The mapping from HUB75 pins to J600/J601 pins is documented in [doc/hub75b_hat.md](../tree/master/doc/hub75b_hat.md)
1. The mapping from J600/J601 pins to FPGA pins is outlined in [doc/hardware.md](../tree/master/doc/hardware.md)
1. Other FPGA pins are described there as well
1. The mapping from the GbE ports (PHY0/PHY1) to the FPGA pins is outlined there as well.
1. So the one thing you are looking for probably (me too, BTW), is a documentation of the description of the network protocol?

jburgess777, 2020-01-22:
> So the one thing you are looking for probably (me too, BTW), is a documentation of the description of the network protocol?

This may help but I haven’t had a chance to try it myself yet.
https://github.com/FalconChristmas/fpp/blob/master/src/channeloutput/ColorLight-5a-75.cpp

tomverbeure, 2020-03-19:
> This is a great work, guys! But it'll be even greater if the original LED matrix driving logic in FPGA can be reverse-engineered and shared here! I'm currently working on a project to develop a FPGA LED matrix display driver using Xilinx LX9. Would love to have some reference codes handy.

Before closing this issue, allow me to add a shameless link to the [RTL](https://github.com/tomverbeure/cube/tree/master/spinal/src/main/scala/cube) for the FPGA that drives [my LED cube](https://github.com/tomverbeure/cube/blob/master/README.md). My code is written is SpinalHDL, but there should be plenty of Verilog implementations for HUB75 drivers.

closed 2020-03-19 by tomverbeure

Kamal-Sonani, 2021-11-24:
@MarcelWaldvogel Yes, I'm on 6th line ,looking for Doc. to turn on firs led with RGB value no matter which language used on system )p