q3k/chubby75 #24: Colorlight 5A-75B project?

jeanthom, 2020-01-02:
Hi!

I saw [this](https://twitter.com/FauthNiklas/status/1212510451365535746) post on Twitter showing a HUB75 FPGA board with an ECP5 instead of a Spartan-6. I plan on reverse engineering it, and I think many other people would be interested in joining the venture.

Should we publish our RE investigations here or create another repo instead?

tomverbeure, 2020-01-09:
Since this board is very similar to the RV901T, I think it makes sense to add the information to this project.

I just merged pull request #27 which adds a directory for 5A-75B information and the JTAG connections that I just figured out.

Eventually, we'll probably need to reorganize, but at least there is placeholder directory now to gather information.

closed 2020-01-09 by tomverbeure

jeanthom, 2020-01-09:
Nice work, can't wait to receive mine!