q3k/chubby75 #26: Is there a readme on how to get this project working?

PM1001, 2020-01-03:
I have been trying in vain to get this project working. It seems there is always an issue with the litex dependencies.

For instance, this is one of the problems I am having:
ImportError: No module named 'liteeth.phy.s6rgmii'

Can anybody give me a hint to how to install all the litex and migen components for this project? 

Thanks

enjoy-digital, 2020-01-03:
You can install LiteX with:
````
$ wget https://raw.githubusercontent.com/enjoy-digital/litex/master/litex_setup.py
$ chmod +x litex_setup.py
$ ./litex_setup.py init install --user (--user to install to user directory)
````

PM1001, 2020-01-03:
Thank you very much! That seems to have done the trick!
For some reason none of the permutations of the installations under third-party/ was working for me.

NiklasFauth, 2020-03-21:
Some more documentation can also be found here:
https://github.com/NiklasFauth/colorlight-led-cube

closed 2020-05-16 by PM1001