q3k/chubby75 #32: Colorlight 5A-75B reverse comment/questions

enjoy-digital, 2020-01-22:
Hello,

thanks a lot @miek@ @smunaut for the reverve, i've been using the reversed pinout to prepare LiteX-Boards support (https://github.com/litex-hub/litex-boards/commit/1d9e34909397f29e69d8dcdab7088cd8b6765a34) and have a few comment/questions:

V6.1: @smunaut
SDRAM: D9 is listed twice, DQ[7] is probably a typo, is it D8?
Gigabit PHYs: maybe you should add the XTALO frequency: 25MHz and that it can be used as the main FPGA clock.
Have you been able to document the led and button? (sorry i can't do it since i have a V7.0)

V7.0: @miek
For the Gigabit PHYs 0: P5 is listed twice: RXC and ~RESET.
Similar to v6.1:
- the input clock (P6)) is also from one of the XTALO of the Gigabit PHYs and is 25MHz. 
- the SDRAM ~CS/~LDQM/~UDQM are connected to GND and CKE connnected to 3v3.

Florent

smunaut, 2020-01-22:
Yeah, I had reversed the LED and button, I forgot to transcribe them in the markdown file.
And you're right about the typo, it's indeed D8, made a mistake when transcribing from my notes.

PR opened to fix those.

enjoy-digital, 2020-01-22:
Thanks @smunaut, @miek for the answers and PRs.

closed 2020-01-22 by enjoy-digital