q3k/chubby75 #37: 5a-75B turning outputs into inputs? also 5a-75e

fanoush, 2020-01-25:
Wondered how the outputs can be turned into inputs. On the 5a-75B when changing TC74VHC245 direction pin it would probably feed 5V into FPGA? so maybe powering from 3.3V instead of  5V would solve that? Does the board work when powered from 3.3 or can the vcc to TC74VHC245 be changed to 3.3V

Thinking about interfacing it with 5V CPU bus like the C64 cartridge connector. 

BTW there is also 5a-75e https://www.aliexpress.com/item/32836510389.html having same Lattice FPGA and twice the connectors, also some listings like this one https://www.aliexpress.com/item/33010384708.html say "Wide working voltage range with DC3.3~6V." so maybe the output of that one is not 5V?

cyber-murmel, 2020-01-30:
> A parasitic diode is formed between the bus and VCC terminals.Therefore, the VHC245 cannot be used to interface 5 V to 3 Vsystems directly.

 [Datasheet](https://www.onsemi.com/pub/Collateral/MC74VHC245-D.PDF)

Although there is a 33 Ohm resistor between the connector and the IC, I wouldn't really trust it to not put 5V on my 3.3V rail.

cyber-murmel, 2020-01-30:
My current plan is to have some PCBs produced to replace the IC and just bridge the connection.

[image]

FFY00, 2020-07-10:
@cyber-murmel do you have the design somewhere? I was thinking of ordering some pcbs.

cyber-murmel, 2020-07-11:
@FFY00 The problem with this particular design was that fabs generally don't accept castellated holes that small.

But I recently got around to design and install another version of this idea. This time with three plated through holes per pad to wick in / wick through the solder.
[image]

FFY00, 2020-07-11:
Hum, that might be a little bit difficult to line up. Let me know when you give it a try. Right now I think I can live with hand soldered jumpers on one or two chips.

cyber-murmel, 2020-07-11:
Most of the holes aren't clogged from the fab. I found it easy to align them by looking through the holes and having the pads of the footprint behind it create a parallax effect. Like when you line up a stencil with a PCB.

cyber-murmel, 2020-07-11:
I also found, that a big single-flat soldering tip on powerful iron with lots of flux is easier to use, than hot air.

tomverbeure, 2020-07-11:
@cyber-murmel Hey cyber-murmel, 
do you mind if I use that picture that you posted above in a presentation about FPGA board hacking?
Tom

cyber-murmel, 2020-07-12:
@tomverbeure The picture above is under the [Creative Commons Attribution-NonCommercial License](https://creativecommons.org/licenses/by-nc/4.0/legalcode)
[image] [image] [image]

Disasm, 2020-08-30:
I came up with these flex boards from OSH Park: https://oshpark.com/shared_projects/Og8YtGBU
With minor rework they can be soldered with a regular sodering iron and some flux.

[image]
[image]

derekmulcahy, 2020-08-30:
Sweet, just ordered a set, thanks.

FFY00, 2020-08-30:
Also ordered, many thanks :blush:!

fanoush, 2020-08-30:
Thanks, I guess one just needs to not forget clicking "Flex" checkbox on ordering page (?)

Disasm, 2020-08-30:
@fanoush yes! Non-flex boards will be useless.

tomverbeure, 2020-08-30:
Nice! Just ordered a bunch of them as well.

jadafi, 2020-09-22:
So just to undestand this correctly: with these bridges the inputs are directly wired to the fpga, thus they are only 3.3V tolerant?

Disasm, 2020-09-22:
@jadafi Exactly.

kittennbfive, 2020-09-23:
I might have found a way to get 5V tolerant inputs. I documented it on my github: https://github.com/kittennbfive/5A-75B-tools/tree/master/I_want_inputs

TL;DR: Replace (some of) the buffers with SN74LVC245APWR with some surgery to modify 2 connections on each IC.

Disasm, 2020-09-23:
Speaking of IC replacement hacks, there is another option that involves SN74CBT3245APW bidirectional FET switches: https://twitter.com/Claude1079/status/1231194849350647808
Seems like it's also requires additional hacks, but looks relatively easy.

eibach, 2022-05-29:
I have documented my modifications here: https://zeromips.org/posts/2022-05-29-5a-75b
Involves flex PCB as well as SN74CBT3245APW.