q3k/chubby75 #39: What is the part number for JTAG

KiranKanchi, 2020-01-26:
Hi,

What is the part number for the USB to JTAG adapter used here ?

Thanks
Kind regards
Kiran Kanchi

jeanthom, 2020-01-26:
Hi Kiran,

You can use any OpenOCD-compatible JTAG adapter. I had lots of success with [LambdaConcept's FT4232H-based adapter](https://shop.lambdaconcept.com/home/25-jtagserial-pack.html) and [Zoobab's fork of Versaloon firmware for STM32 MCU](https://github.com/zoobab/versaloon) which is a cheaper yet slower alternative.

tomverbeure, 2020-01-27:
I've been successful using a CPLD-based Altera USB-Blaster compatible JTAG programmer.

Since open source tools are often programmer-agnostic while commercial FPGA software from Xilinx and Altera are not, I recommend one that is Xilinx or Altera compatible.

KiranKanchi, 2020-01-27:
Thank you very much for your kind reply.

Best regards
KiranKanchi

closed 2020-01-27 by KiranKanchi

eparadis, 2021-02-04:
> I've been successful using a CPLD-based Altera USB-Blaster compatible JTAG programmer.

Do you have any further info about this? I have an Altera USB-Blaster clone that came with a simple MAX2 breakout board. Using `openocd` results in a `tdo check error` when I try to program [the hello world example here](https://github.com/kittennbfive/5A-75B-tools/tree/master/Hello_World_1).

The command I'm using is 
`openocd -f /usr/local/share/openocd/scripts/interface/altera-usb-blaster.cfg -f prog_openocd.cfg`

The wiring I'm using is 
```
(v7.0 board)
board - name - USB Blaster pin
J27 - TCK - 1
J31 - TMS - 5
J32 - TDI - 9
J30 - TDO - 3
J33 - 3.3v - 4 (VccTRGT)
J34 - GND - 2
```