q3k/chubby75 #43: More infomation about using OpenOCD to connect to the board

oloftangrot, 2020-02-26:
Is it possible to add some more information about how to use OpenOCD?
Like configuration files and some basic command's what ever on how to verify connection and deply a design on the fpga.

cyber-murmel, 2020-02-26:
Maybe I'll write something down, but for now, that's what has worked for me to upload a bitstream.
`chubby.cfg`
```
transport select jtag
adapter_khz 500

#jtag newtap auto0 tap -irlen 2 -expected-id 0x41111043
jtag newtap lfe5u25 tap -expected-id 0x41111043 -irlen 8 -irmask 0xFF -ircapture 0x5

init
scan_chain
svf -tap lfe5u25.tap -quiet -progress top.svf
shutdown
```
I used a J-Link
```bash
openocd -f interface/jlink.cfg -f chubby.cfg
```

oloftangrot, 2020-03-04:
Thanks, I have not been able to test this since I got V7 HW and the top.lpf contains pin allocations, or whatever, that does not fit the fgpa on the board.

oloftangrot, 2020-03-06:
I use a usbblaster (imported from China) of unkkonw quality but it seems like I get some error:

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Warn : Adapter driver 'usb_blaster' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
Info : No lowlevel driver configured, will try them all
Info : usb blaster interface using libftdi
Info : This adapter doesn't support configurable speed
Info : JTAG tap: lfe5u25.tap tap/device found: 0x41111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x4)
Error: lfe5u25.tap: IR capture error; saw 0x1 not 0x5
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined
   TapName             Enabled  IdCode     Expected   IrLen IrCap IrMask
-- ------------------- -------- ---------- ---------- ----- ----- ------
 0 lfe5u25.tap            Y     0x41111043 0x41111043     8 0x05  0xff
shutdown command invoked

Can it be related to the used package on the v7.0 hardware?

Downloading top.svf fails like this:

Info : This adapter doesn't support configurable speed
Info : JTAG tap: lfe5u25.tap tap/device found: 0x41111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x4)
Error: lfe5u25.tap: IR capture error; saw 0x1 not 0x5
Warn : Bypassing JTAG setup events due to errors
Warn : gdb services need one or more targets defined
   TapName             Enabled  IdCode     Expected   IrLen IrCap IrMask
-- ------------------- -------- ---------- ---------- ----- ----- ------
 0 lfe5u25.tap            Y     0x41111043 0x41111043     8 0x05  0xff
svf processing file: "top.svf"
85%    Error: tdo check error at line 28
Error:     READ = 0x5e01e10
Error:     WANT = 0x0000000
Error:     MASK = 0x000b000
Error: fail to run command at line 12510
Error: tdo check error at line 28
Error:     READ = 0x5e01e10
Error:     WANT = 0x0000000
Error:     MASK = 0x000b000

Time used: 0m1s864ms 
svf file programmed failed

Any suggestions on how to fix this is much appreciated.

rjeschmi, 2020-03-18:
Did you get it to work? FWIW I used the auto detect tap I got and just made a modification on the irlen.

I'm not sure the irlen is right yet.

so my line looks like:
`jtag newtap lfe5u25 tap -irlen 8 -expected-id 0x41111043`

oloftangrot, 2020-03-18:
Programming using the above line fails the same way. I suspect the issue could  be caused by 5 V/3V level incompatibility. I have tried a JLink-clone with even worse result. But that could likely due to some interface firmware issue. So right now I don't have a JTAG interface I really trust. I guess I could try to add level shifters to the usbblaser interface but since I not sure that is the problem or how to do it in a practical way it is not what I want to spend time on.

rjeschmi, 2020-03-19:
I'm not an expert in this at all, but I did manage to get it working (I think). I wrote something to it anyway, it errorred for other reasons until I replaced my usb cable (the other one was kind of long. I'm using a bus blaster, which emulates one of the standard ft2232 devices.

This stuff should all be 3.3V. You will need 5V to the board (I assume. I powered the board and gave the 3.3 back to the bus blaster as a reference).

You might have to mess with the jlink.conf file. I pulled the info out of the dp_busblaster.conf and just put it in one openocd file for  testing.

```
interface ftdi
ftdi_device_desc "Dual RS232-HS"
ftdi_vid_pid 0x0403 0x6010

ftdi_layout_init 0x0c08 0x0f1b
ftdi_layout_signal nTRST -data 0x0100 -noe 0x0400
ftdi_layout_signal nSRST -data 0x0200 -noe 0x0800

ftdi_channel 0

transport select jtag
adapter_khz 50

jtag newtap lfe5u25 tap -irlen 8 -expected-id 0x41111043
#jtag newtap lfe5u25 tap -expected-id 0x41111043 -irlen 8 -irmask 0xFF -ircapture 0x5

init
scan_chain

svf -tap lfe5u25.tap -quiet -progress top.svf
#shutdown 
```
I commented out the shutdown because it seems to cause an exit before I see more confirmation that the code uploaded.

the last error (`85% Error: tdo check error at line 28`) that I got a lot like yours was cause by my USB cable being too long and error prone (I don't think there is any checking of the CRC to make sure the sent data is valid. This seems to be a kind of common problem with these devices.

and then I run it with:

```
(base) root@node1:~/busblaster# openocd -f dp_busblaster.cfg

Open On-Chip Debugger 0.10.0+dev-01012-ged8fa09cf-dirty (2020-01-15-04:14)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : clock speed 50 kHz
Info : JTAG tap: lfe5u25.tap tap/device found: 0x41111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111,
ver: 0x4)
Warn : gdb services need one or more targets defined
svf processing file: "top.svf"
85%    Warn : Haven't made progress in mpsse_flush() for 2002ms.
Warn : Haven't made progress in mpsse_flush() for 2002ms.
Warn : Haven't made progress in mpsse_flush() for 2002ms.
Warn : Haven't made progress in mpsse_flush() for 2003ms.

<clip>
```

oloftangrot, 2020-03-20:
Ok, thanks for all information. I have not connected the 3 V from the board to the Blaster interface since it was not clear to me which pin to use and it sort of worked anyway. Which pin did you use?

oloftangrot, 2020-03-21:
So I have tested to connect the 3.3 V board pin to interface pin 4 with about the same result as before.

Trying the direct ftdi driver as suggested above results in a new interface error.
Error: unsupported FTDI chip type: 0x0400
Error: unable to open ftdi device with vid 09fb, pid 6001, description 'USB-Blaster', serial '*' at bus location '*'

Anyway thanks for the suggestions. My interface might have one of those FTDI-chip clones that FTDI  bricked in a Windows driver update some years ago. It seems to work better with the OpenOCD build in driver for usbblaster (I have not found the cfg-file for it) or mayby the later problems with the programming is cased by some assumption about the hardware in the drivercode that this chip is not compliant with.

I placed an order for a JLINK Edu from Germany so hopefully I will have better luck with it when it arrives.

jeanthom, 2020-03-21:
I wouldn't trust USB Blaster clones since they have a wide variety of chips inside (some have seen STM32F103, mine has a PIC18F14K50). Anyway I haven't managed to get mine working.

If you have a Blue Pill (STM32F103) development board lying around you can try [Versaloon](https://github.com/zoobab/versaloon). It worked fine for flashing the gateware on my RV901T board (Xilinx based) albeit quite slow.

**Shameless plug:** if you have any STM32F1 hardware and some free time, please help me add DirtyJTAG support to OpenOCD ([repo here](https://github.com/jeanthom/openocd-dirtyjtag)) :smiley:

oloftangrot, 2020-03-21:
Yes there are some blue pills laying around... I checked my usbblasters and they all (2) uses PIC  which seems to emulate the FTDI good enough for some Quartus dev tool I tested some years back but not compliant with the Ubuntu OpenOCD drivers. Nor did they seem to have the appropriate level conversion interface just a plain 74HC device. While the JLINK clone in contrary looked like high quality on the interface side.

While eager to make some progress I erased the JLINK clone and reflashed it with a firmware I found on internet. I connected the 3.3 V this time and finally the download works without faults using the following cfg.

transport select jtag
adapter_khz 250

jtag newtap lfe5u25 tap -expected-id 0x41111043 -irlen 8 -irmask 0xFF -ircapture 0x5

init
scan_chain
svf -tap lfe5u25.tap -quiet -progress top.svf
shutdown

Actually it could be fun testing Versaloon or help with DirtyJTAG but I guess I will try to  stay focused on the FPGA learning path instead.

jeanthom, 2020-03-21:
Do you mind sending me the firmware you used for J-LINK (my email is available on my GitHub profile)? I have one at home and I can't get it to work properly with OpenOCD.

Also, did you encounter the `IR capture error` error message with your J-LINK probe like everybody here seems to?

oloftangrot, 2020-03-21:
Version 8 firmware was downloaded from this place:
https://gronlier.fr/blog/2015/07/unbrick-and-update-an-j-link-v8-clone/v8_firmware_noserial_crackn/

Instruction followed:
https://gronlier.fr/blog/2015/07/unbrick-and-update-an-j-link-v8-clone/

Used SAM-BA 2.18:
https://www.microchip.com/developmenttools/ProductDetails/PartNO/SAM-BA%20In-system%20Programmer

Don't recall the if the capture error persisted. Will verify later.

oloftangrot, 2020-03-22:
OpenOCD log from run:

Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 250 kHz
Info : No device selected, using first device.
Info : J-Link ARM V8 compiled May 27 2009 17:31:22
Info : Hardware version: 8.00
Info : VTarget = 3.319 V
Info : clock speed 250 kHz
Info : JTAG tap: lfe5u25.tap tap/device found: 0x41111043 (mfg: 0x021 (Lattice Semi.), part: 0x1111, ver: 0x4)
Warn : gdb services need one or more targets defined
   TapName             Enabled  IdCode     Expected   IrLen IrCap IrMask
-- ------------------- -------- ---------- ---------- ----- ----- ------
 0 lfe5u25.tap            Y     0x41111043 0x41111043     8 0x05  0xff
svf processing file: "top.svf"
95%    
Time used: 0m20s782ms 
svf file programmed successfully for 620 commands with 0 errors
shutdown command invoked

oloftangrot, 2020-03-24:
I just wanted to confirm that the JLink EDU also works fine with the above setting. Also tested at 500 kHz with no errors.

closed 2020-03-24 by oloftangrot