q3k/chubby75 #57: SDRAM controller demo?

oloftangrot, 2020-04-11:
Does anyone have a working SDRAM controller example for the ESMT devices used on V7 hardware? I quickly looked at the 3'rd party LiteX i the repos but it seems like an uncertain way to start there.

r4d10n, 2020-04-12:
From https://twitter.com/BitlogIT/status/1244738826431578113 :

https://gist.github.com/GuzTech/6739255f45bdc8394df5db4c7b4a272f

oloftangrot, 2020-04-12:
Tnks for the response and the pointers. LiteX still seems a bit to big in scope for me at the present. Pulling in all these other build and configuration tools apart from those already required in the FPGA toolchain for so many new problems and learning thresholds. Though using ethernet on the board is some kind of a dream goal for me. 

But I putting up LiteDRAM for some trial mayby could be a start then.

Does anyone have some pointers in that direction or any other advice?

oloftangrot, 2020-08-26:
The  MT48LC1M16A1 used at fpga4fun.com seems very similar according to the data sheet. Same 512x16x2 bank organization, same pin out, the command structure seems to be the same. Are these parts equvivalent or can be used in an equvivalent mode?

kittennbfive, 2021-07-12:
Sorry to push this, has anybody made some progress regarding the SDRAM? I already spent hours and days trying to write a controller (for 5A-75B V7 specifically) but it is too hard for a beginner and i don't have the tools to debug (in simulation everything looks fine as far as i can tell). I think if we had a working and FOSS SDRAM-controller maybe we could make some progress in using these boards. 
If somebody really want to look at my current code i can upload it, but it does not work and has been written by a beginner, so quality is probably horrible...

LearnShareAlways, 2023-04-22:
I have V8 with M12L64322A  and it was not working. I had to fix the target board which is using M12616161A  for v8(as you can see v8 goes into else path where M12616161A is being used which is wrong): 
[image]