q3k/chubby75 #61: novastar mrv366?

mpictor, 2020-06-02:
Hi, saw this 16 port card and was wondering if anyone knew what fpga/memory was on it, its pinout, etc. It has a microcontroller and fpga according to the PDF, and apparently firmware updates for both can be done over the network with something called NovaLCT.

firmware and specs are available if you scroll down [here](https://www.novastar.tech/downloads/receving-card/)

direct links -
[specs](https://oss.novastar.tech/uploads/2019/11/MRV366-Receiving-Card-Specifications-V1.2.0.pdf)
[firrmware zip](https://oss.novastar.tech/uploads/2020/04/DATA_MRV366_V4.6.2.0.zip)
I also saw [older specs](https://www.ledcontrollercard.com/media/wysiwyg/Novastar/Novastar%20MRV366%20Receiving%20Card%20with%2016%20HUB75%20ports%20Description.pdf) on another site.

Note that all firmwares seems to be compressed in an unknown way, or (more likely) encrypted. All I get out of binwalk is that the entropy is very high. There don't seem to be any null sequences in the files, and the sizes are evenly divisible by 2k.

The spec sheet seems to indicate these may be driven directly by software, rather than a dedicated sender card, so I wonder if it contains more memory than the other receiver cards.

tomverbeure, 2020-06-02:
The [product photo on Amazon](https://www.amazon.com/Novastar-MRV366-Receiving-Display-512x256/dp/B07GNJ8M4C) shows a DRAM number that ends with 322A.

When you go to [the page on the ESMT website](https://www.esmt.com.tw/en/Products/DRAM/SDRAM-1-1) that has all the DRAMs, there is one SKU with those letters: M12L64322A(2S), which has a capacity of 2Mx32. There is only one of those on the board.

The RV901T also has a single 2Mx32. The Colorlight 5A-75B has 2 SDRAMs with a capacity of 1Mx16 each.

So that settles the memory question.

As for the FPGA: there's only one way to find out. :-)

Bradyok, 2020-06-08:
If anyone wants to work on this, I have 100s of these NovaStar cards in stock, contact me brady@olympianled.com

tomverbeure, 2020-06-08:
Brad, 
Could you check on one of those cards if what kind of FPGA is on there?
Tom

Bradyok, 2020-06-09:
The writing on the FPGA:

XC6SLX16
CSG324DIV1929
D5953437A

tomverbeure, 2020-06-09:
Thanks!

That makes the card very similar to the RV901T. 

Personally, I’m more interested in the ECP5 FPGAs because the availability of an open source flow, and I assume many others will be in the same boat.

So I’m going to give this one a pass.

Duality4Y, 2023-01-13:
I have such a board right in front of me there is a Spartan-6 on there yosys supports this chip.
And I believe nextpnr has spartan-6 support as well.