q3k/chubby75 #65: 5A-75E V6.0

lichtfeind, 2020-08-01:
I ordered some colorlight 5A-75E and got one with V6.0.

The Chips used:
 * 1x `LFE5U-25F 6BG381C`
 * 1x `25Q16CSIG` Flash
 * 2x `EM636165TS-6G` RAM
 * 2x `850612D` Ethernet
 * 20x `VHC245 3600097`

I have not used it so far, so no pinout so far. And they might be no longer be made so not sure how imported this is.

[image]
[image]

Disasm, 2020-08-30:
Apparently, now we have two completely different 5A-75E V6.0 in the wild. Another one was documented recently by @adamgreig: https://github.com/q3k/chubby75/pull/67

adamgreig, 2020-08-31:
Huh, that's super weird. It actually looks a lot closer to the V7.1 hardware than my V6.0; notably it also has two 16-bit RAM chips and uses the same Broadcom PHYs as V7.1. However it has the 381-ball package for the ECP5 instead of the 256-ball that's on my 6.0 and the other 7.1 hardware. I wonder if it's mean to be a sort of "7.0". Could you say when you purchased this, @lichtfeind ? It would also be interesting to know if the bitstream has any comments at the start.

lichtfeind, 2020-09-02:
I ordered two 5A-75E on 2020-07-18 an got one V7.1 and one V6.0 in the same package. i don't have a Flash dump of the V6.0 atm. You will have to wait until the weekend.

adamgreig, 2020-09-02:
Thanks for the details and no rush on the bitstream, I'm just curious!

Disasm, 2020-09-02:
Look, someone is collecting flash images for science! :D https://github.com/suglover/5a-75x-images

lichtfeind, 2020-09-04:
@adamgreig yes it looks like there are some infos.

V6.0 6BG381C:
```
Lattice Semiconductor Corporation Bitstream
Version:         Diamond (64-bit) 3.8.0.64.0
Bitstream Status: Final Version 10.23
Design name: lattice_lhf_lattice_lhf.ncd
Architecture: sa5p00
Part: LFE5U-25F-6CABGA381
Date: Wed Mar 07 12:05:28 2018
Rows: 7562
Cols: 592
Bits: 4476704
Readback:     Off
Security:     Off
Bitstream CRC: 0x3B81
```

And that makes the bitstream two month older then the V7.1: `Thu May 10 10:37:38 2018`

Disasm, 2020-09-04:
@lichtfeind *2018* :)

lichtfeind, 2020-09-04:
Yes the image at https://github.com/suglover/5a-75x-images would support that.

I don't know how licensing for diamond works but 3.8 looks quit old. So maybe they changed the clock.

I also got a V7.1 board with 2MB Flash and that has comments: (The 4MB Board does not)
```
Lattice Semiconductor Corporation Bitstream
Version:         Diamond (64-bit) 3.8.0.115.3
Bitstream Status: Final Version 10.23
Design name: lattice_lhf_lattice_lhf.ncd
Architecture: sa5p00
Part: LFE5U-25F-6CABGA381
Date: Thu May 10 10:37:38 2018
Rows: 7562
Cols: 592
Bits: 4476704
Readback:     Off
Security:     Off
Bitstream CRC: 0x3B81
```
So i am not sure if the comments are even related to the bitstream as that is not an 381-ball part but a 256.
I think there are multiple configs in that flash so this might be just for some old boot loader.

Disasm, 2020-09-04:
Huh. I think it's possible to build a bitstream for 381-pin package that will work for 256-bin package, but this case is strange anyway.

derekmulcahy, 2020-09-04:
I built the pin_scan bitstream and the 381-pin bitstream worked when up
uploaded to the 256-pin device. The reported PIN numbers were wrong but it did work.