q3k/chubby75 #73: 5A-75E v8.0?

DoctorMikeReddy, 2020-12-27:
I’ve just acquired a v8.0 5A-75E board. It’s not mentioned here, and the JTAG (?) connectors are a square now, not a row. Will try to analyse over the next few days.
[image]
[image]

DoctorMikeReddy, 2020-12-29:
Well, I’m not imagining it…
[https://www.lednets.com/upload/Support/5A-75E-Specification%C2%A0V8.0091586.pdf](https://www.lednets.com/upload/Support/5A-75E-Specification%C2%A0V8.0091586.pdf)

kargeor, 2021-06-05:
Did you get it to work? I got the same one and I cannot get the JTAG to work.

DoctorMikeReddy, 2021-06-05:
Mine has been sitting in storage for months :-( as I had to move house a couple of times; divorce sucks. I’m planning to return to this over the Summer

Disasm, 2021-08-08:
@kargeor check the JTAG pinout for 5A-75E V6.0, it should be the same.

kargeor, 2021-08-08:
I got it to work. Thanks.

DoctorMikeReddy, 2021-08-08:
Oh, what did you have to do to get it working? Anything special?

kargeor, 2021-08-09:
https://github.com/trabucayre/openFPGALoader/issues/88
I needed to add a resistor to TDO.

cdwijs, 2022-10-19:
I also got it to work. Refer to the last message on this forum:
https://forum.linuxcnc.org/27-driver-boards/47028-my-notes-flashing-litex-cnc-on-a-colorlight-5a-75e-solved?start=10
[root@cedric cedric]# pacman -S openfpgaloader
[cedric@cedric ~]$ openFPGALoader -c usb-blaster -m LiteX-CNC/5a-75e/gateware/colorlight_5a_75e.bit
write to ram
Open file: DONE
Parse file: DONE
Enable configuration: DONE
SRAM erase: DONE
Loading: 100.00%
Done
Disable configuration: DONE
[cedric@cedric ~]$

roby2014, 2022-12-16:
Has anyone managed to use a FTDI232 to program the board? I connected all the pins (like the above comment), but however openFPGALoader is not detecting the board.
```
❯ openFPGALoader --detect
unable to open ftdi device: -3 (device not found)
JTAG init failed with: unable to open ftdi device
```

EhsanRasouli1371, 2024-01-01:
Hi, I just bought this board and I don't know much about how to configure the ini and hal files
Can you send me your configuration files?
Thanks