q3k/chubby75 #85: RV901T inputs 5V

mmueller-kaffeeschluerfercom, 2021-09-01:
For RV901T, the I/O direction of some of the 74HC245 buffers can be set by the FPGA on pin F13 (U600-U607, see [Section Buffer](https://github.com/q3k/chubby75/blob/master/rv901t/doc/hardware.md#buffers). But the VDD on U600-U607 is at 5 V and like also mentioned in #37 for the 5a-75B, the 74HC245 buffers drive the FPGA inputs at 5 V when configured as an input buffer (the 74HC245 is directly connected to the FPGA pins). According to [Xilinx](https://forums.xilinx.com/t5/Spartan-Family-FPGAs-Archived/Is-Spartan-6-Inputs-5V-tolerant/td-p/427546), the Spartan-6 is not 5 V tolerant, meaning that it would damage or destroy the FPGA if the buffers are configured as inputs.

Am I missing something or is the board really designed that way?

Shouldn't this be mentioned in the Hardware description?

At least, the DIR input of the buffers is pulled to GND by R608, so during configuration of the FPGA, the buffers are configured as outputs and likely stay like this if the new design does not configure F13.

pauluzs, 2021-09-04:
If you don't need the 5v output level and the 5v tolerant input level try to run the board on 3.3V
This works for me when feeding 3.3V into the 5v line which also feeds the 3.3v regulator.
Or you could also try and feed the 3.3V directly on JP2

mmueller-kaffeeschluerfercom, 2021-09-06:
Thanks a lot, great to know the board workds with 3.3 V!

But do you agree with me it's a bad idea to put F13 to high when powered with 5 V? If yes, I would propose to mention it in the hardware description.

pauluzs, 2021-09-06:
According to linsn's [datasheet](https://www.linsnled.com/download/RV901_911_921_931%20Specification.pdf) the board should work between 3.3 and 5.5v

And according to the 74HC245 [datasheet](https://assets.nexperia.com/documents/data-sheet/74HC_HCT245.pdf) the buffers should work between 2.0 and 6.0 V

Been running it as a mesa 7i90 card with hostmot2 firmware over spi to  linuxcnc on a raspberry pi 4 for several months without any problems. Albeit  still using a  5V > 3.3V level converter on the inputs. 

Now you brought this up i wonder if the inputs will be 5V tolerant when the board and buffers are fed from 3.3V?

mmueller-kaffeeschluerfercom, 2021-09-06:
If I understand correctly, the 74HC245 datasheet, section 7, table 4 (absolute maximum rating), states that the input clamping current must not exceed 20 mA (at condition VI > VCC + 0.5 V). So, assuming the clamping diode has a voltage drop of 0.5 V, the remaining (5 - 0.5 - 3.3) = 1.2 V drops in the current limiting resistor, which is 75 Ohm on RV901T, so there would be an input current of 16 mA which is 80 % of absolute maximum rating. Dirty hack but may work. So at least in theory, at VCC = 3.3 V, the 74HC245 inputs should be 5 V tolerant with a current limiting resistor of 75 Ohm.

Bleeptech, 2021-11-07:
"Been running it as a mesa 7i90 card with hostmot2 firmware over spi to linuxcnc on a raspberry pi 4 for several months without any problems."

Do tell! Details man, details! LOL

Peter-van-Tol, 2022-04-21:
pauluzs commented...

"Been running it as a mesa 7i90 card with hostmot2 firmware over spi to linuxcnc on a raspberry pi 4 for several months without any problems."

I'll second that: definitely want to know more. Do you have any links on how you flashed the firmware to the card? Is it by coincidence that the Mesa and the RV901T uses the same chipset?

On the other side: thanks to the great work in this repository there is also a development to create a dedicated firmware and driver for LinuxCNC which works on ethernet: https://github.com/Peter-van-Tol/LiteX-CNC/tree/main (still work in progress).

Will try to run the board at 3.3 Volt and modify the buffers to fix in and outputs.

pauluzs, 2022-04-21:
Sorry for the late reply, 
Didn't see bleeptech's comment.

It's been a while so i had to search my backups to see what i could find.
In short i made a custom .ucf and .vhd for the rv901's fpga
[RV901Tspi.ucf.txt](https://github.com/q3k/chubby75/files/8535029/RV901Tspi.ucf.txt)
[RV901T_x16card.vhd.txt](https://github.com/q3k/chubby75/files/8535030/RV901T_x16card.vhd.txt)

Flashing was done by directly flashing the bitstream to flash memory, with openocd on the raspberry pi connected to JP5 , no soldering needed standard dupont jumpers will fit. The linuxcnc spi>fpga connection was made at jp4 which is 3.3V (needs pins soldered)

I did find some notes i made when i did this , this includes the whole process, for the fpga flashing see line 180 and down
[pi4_5.9_rt_linuxcnc.txt](https://github.com/q3k/chubby75/files/8535096/pi4_5.9_rt_linuxcnc.txt)

Hope this will help you out

Br Pauluzs

tomverbeure, 2022-11-21:
@ericfont 
Eric, 
I have merged your pull request. Do you think this is sufficient to close this issue? 
Tom

ericfont, 2022-11-21:
yes, I think this can be closed now, thanks.