q3k/chubby75 #86: RV901T: Errors when building LiteX examples

mmueller-kaffeeschluerfercom, 2021-09-01:
Just installed ISE 14.7 and LiteX as described [here](https://github.com/enjoy-digital/litex). I can successfully build the blink example in ISE and configure the FPGA. But when trying to build sdram_test.py, I get following error:

`litex/riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: bios.elf section ``.rodata' will not fit in region ``rom'
litex/riscv64-unknown-elf-gcc-8.3.0-2019.08.0-x86_64-linux-ubuntu14/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: region ``rom' overflowed by 840 bytes
collect2: error: ld returned 1 exit status
`

and when trying to build rgmii_test.py

`Elaborating module <PLL_ADV(BANDWIDTH="OPTIMIZED",CLKFBOUT_MULT=6'b100101,CLKFBOUT_PHASE=0.0,CLKIN1_PERIOD=40.0,CLKIN2_PERIOD=0.0,CLKOUT0_DIVIDE=3'b111,CLKOUT0_DUTY_CYCLE=0.5,CLKOUT0_PHASE=0.0,CLK_FEEDBACK="CLKFBOUT",COMPENSATION="INTERNAL",DIVCLK_DIVIDE=1'b1,REF_JITTER=0.01,SIM_DEVICE="SPARTAN6")>.
ERROR:HDLCompiler:267 - "/home/user/fpga/chubby75/rv901t/build/gateware/platform.v" Line 5398: Cannot find port PWRDWN on this module`

Has anyone an idea of what I am doing wrong?

enjoy-digital, 2021-09-01:
Hi @mmueller-kaffeeschluerfercom,

this is probably an issue in LiteX that has been fixed with https://github.com/enjoy-digital/litex/pull/962. Updating LiteX should  fix it.

mmueller-kaffeeschluerfercom, 2021-09-01:
Ah, yes, building rgmii_test.py now works (btw, I had to comment out the line, `./litex_setup.py update` did not update the file `litex/soc/cores/clock/xilinx_s6.py`)<br> Thanks a lot!

The problem with sdram_test.py persists, though.

enjoy-digital, 2021-09-01:
The size of the BIOS probably increased a little bit. You can just remove the `integrated_rom_size` configuration here: https://github.com/q3k/chubby75/blob/master/rv901t/sdram_test.py#L60 since LiteX now automatically reduces it or keep it and increase it.

mmueller-kaffeeschluerfercom, 2021-09-01:
Yes, thank you! One step further :-)

Unfortunately, now, I get another error:

`Phase 9.8  Global Placement`
`ERROR:Place:543 - This design does not fit into the number of slices available
`
...
`Phase 11.9  Local Placement Optimization`
`ERROR:Place:543 - This design does not fit into the number of slices available`
...
`ERROR:Place:120 - There were not enough sites to place all selected components.
`
...
`ERROR:Pack:1654 - The timing-driven placement phase encountered an error.`

I tried `export XIL_PAR_ENABLE_LEGALIZER=1` but with same errors.

enjoy-digital, 2021-09-02:
It's possible the resource usage increased a little bit and that the design is now too big.

With:
```python3
        SoCSDRAM.__init__(self, platform,
            cpu_type            = "vexriscv",
            cpu_variant         = "minimal",
            clk_freq            = sys_clk_freq,
            integrated_rom_size = 0x8000)
```
it's building fine. 

Your first issue is also fixed with https://github.com/enjoy-digital/litex/commit/7bd06d178f33c2bdce28f648e553dcd1d87d7878.

mmueller-kaffeeschluerfercom, 2021-09-03:
excellent! thank you very much!

apahm, 2021-09-08:
@mmueller-kaffeeschluerfercom, rgmii_test.py work? Did you try ping?

mmueller-kaffeeschluerfercom, 2021-09-08:
I just ran a quick test. The ping doesn't work. When I start pinging, I get an broadcast ARP request from the FPGA which is answered from my computer. But then, nothing more. Unfortunately, right now I don't have time to look at this further...

apahm, 2021-09-08:
@mmueller-kaffeeschluerfercom thank you for your reply!

mmueller-kaffeeschluerfercom, 2021-09-13:
Has anyone an actual build with working ping?

enjoy-digital, 2021-09-13:
I've been able to test successfully the LiteX-Boards design: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/linsn_rv901t.py with` ./linsn_rv901t.py --cpu-type=None --integrated-main-ram-size=0x100 --with-etherbone --build` and ping it at `192.168.1.50`.

The issue with `rgmii_test.py` here is probably that the `tx_delay` of the PHY now need to be specified since the default value has changed:
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/linsn_rv901t.py#L78

enjoy-digital, 2021-09-13:
If you still have troubles, the hardware will be on my desk for the next couple of days, so feel free to ask.

mmueller-kaffeeschluerfercom, 2021-09-13:
Yes, this was it: just add `tx_delay   = 0e-9` to the PHY and ping works! Thank you very much!! I'll create a pull request.

closed 2021-09-13 by mmueller-kaffeeschluerfercom