q3k/chubby75 #94: I need help with Colorlight 5A-75E v 6.1

synkalimonad, 2022-05-23:
I need help with Colorlight 5A-75E v 6.1 receiver cards. After installing firmware version 5A 11.04.fw via LED upgrade software into Colorlight 5A-75E v 6.1 receiving cards. Receiving cards start up but the LED is solid green and receiving cards are not displayed in the ledvision and ledupgrade software. How would you advise to return the old firmware and functionality to the receiving cards. Thank you

q3k, 2022-05-31:
I don't think anyone here experimented enough with the original firmware? But I'll leave this open just in case someone has an idea.

kittennbfive, 2022-07-18:
> How would you advise to return the old firmware and functionality to the receiving cards.

You should be able to reflash the external FLASH memory with the original bitstream - if you can get a copy of the bitstream somewhere. Depending on the exact type of flash you might have a chance with flashrom and some hardware (maybe a CH341?).