q3k/chubby75 #100: Gateware to determine connections

cdwijs, 2022-10-22:
I would like to create gateware that shows what ball of the FPGA goes to what point on the headers.

On each pin, I would like to output a train of pulses, that indicate the ball. First a preamble, and then 2 burst indicating the row and column That way I can look at an LED to figure out the connection.

For example:
A2: 10101010101010000010000010010000 (and then repeat)
G4: 10101010101010000010010010010010010010000010010010010000
C4: 10101010101010000010010010000010010010010000

Has this already been done?
What is the best way to proceed?

tomverbeure, 2022-10-22:
The more common approach is to assign each IO pin their own UART that outputs a unique l, fixed, number, and to use an oscilloscope or logic analyzer (such as a Saleae or compatible) that has built-in UART decoding functionality.

It’s not very hard to make that yourself. Chances are that you can find such a design somewhere on the web.

Tom

cdwijs, 2022-10-22:
Where can I find the file that defines where all the connections are in the FPGA?

cdwijs, 2022-10-22:
Here's the part of LiteX that uses the information from this repository:
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/colorlight_5a_75e.py