q3k/chubby75 #101: Inconsistent pinout, J16, 5A-75E V6.0

rinatzakirov, 2022-11-06:
On the 5A-75E V6.0 board J16 has an inconsistent FPGA assignment.
HUB75 pins 1 and 6 are both assigned to FPGA Pin F14.
This is likely a typo.

rinatzakirov, 2022-11-06:
Just confirmed that Pin 6 of J16 is connected to F14, and Pin 1 is something else.

rinatzakirov, 2022-11-06:
Figured it out:
J16, Pin 1 is G14

So the fix is to change 
F14, 1, R0
to:
G14, 1, R0

FPGA Pin | HUB75 Pin | Function
-- | -- | --
F14 | 1 | R0
G13 | 2 | G0
F12 | 3 | B0
F13 | 5 | R1
F14 | 6 | G1
E14 | 7 | B1

FPGA Pin | HUB75 Pin | Function
-- | -- | --
F14 | 1 | R0
G13 | 2 | G0
F12 | 3 | B0
F13 | 5 | R1
F14 | 6 | G1
E14 | 7 | B1

rinatzakirov, 2022-11-06:
Merge request created, along with a few more, hopefully useful, additions.
https://github.com/q3k/chubby75/pull/102

kubabuda, 2022-12-29:
PR is merged, perhaps issue can be closed?

closed 2022-12-29 by tomverbeure