q3k/chubby75 #105: Input for ColorLight 5A-75B/E, by rewiring pin 1 of buffers to and powering with 3.3v?

ericfont, 2022-11-21:
To use inputs for ColorLight 5A-75B/E, I was thinking what about rewiring a buffer IC's pin 1 (the direction pin) to disconnect it from Vdd and instead connect it to ground?  It would require physically breaking pin 1's connection to the board cause the pin sits on Vdd rail.  Then could carefully solder a wire to it and an exposed ground.

Would also need to power the entire board using 3.3V instead of 5.0V, since the FPGA pins are surely not 5V-tolerant. (The only things the positive rail powers are the buck converters, but according the data sheet the 3.3V buck converters can still work with 3.3V input...the output just can't be higher than the input.)

ericfont, 2022-11-21:
The readme currently suggests to replace the buffer boards instead, but I wanted to suggest just physically rewiring pin 1 as a simpler solution (which requires less soldering and doesn't require buying a new chip).

ericfont, 2022-11-21:
I just wanted to note the downside of my suggestion is that once pin 1 has been physically rewired, it would be difficult to go back.  Hypothetically one could instead rewire the pin to a physical switch.  Or could even wire to an output pin from another output-only buffer, to allow that one wire to be selectively reconfigured by writing to that output pin.

kittennbfive, 2022-11-24:
Yes, this should work. However as you said the FPGA is certainly not 5V tolerant and i don't know if the entire board will work when powered with 3,3V (not familiar with this buck converter stuff).

Peter-van-Tol, 2022-11-28:
It is worth to mention that the buffers (`74HC245T`) are not 5 V tolerant either when driving, so the peripherals should be also driven by 3.3 Volts in this case. A slightly less intervening method to convert the board, whilst being able to use 5 Volt inputs is to run the board from 3.3 Volt and replacing the buffers with `74LVS245`. In this case you don't have tow rewire / cut the `VCC` to each chip you replace, making damages less likely to happen. You only have to cut the trace to the direction pin and rewire that to GND.

stanleyseow, 2023-01-08:
Read here https://twitter.com/claude1079/status/1231194849350647808?lang=en