q3k/chubby75 #107: 5A-75E V8.2 cannot connect ethernet

Peter-van-Tol, 2023-01-14:
I've received a new batch of 5A-75E cards. Unfortunately this is a version 8.2 card. This card has the same FPGA and memory as the previous card, however I suspect that the pinout is different: I cannot connect with etherbone with this card. Does anybody else had some more success with a V8.2 card?

On the other hand: How to do pin mapping for PHY?

Peter-van-Tol, 2023-01-15:
Checked the components on the board. The only difference with version 6.0/8.0 seems to be the change of the FPGA to LFE5U-25F-7BS256J. This appears to be a higher speed grade but cannot imagine that this would be a valid reason to change layout of the board itself. The other components are equal.

Peter-van-Tol, 2023-01-15:
This issue can be closed. It must have been a fluke while compiling. Re-compiled and reflashed the board and it now responds to ping messages. 

Conclusion: When you receive a revision 8.2 board, you can use the version 6.0 pinout.

closed 2023-01-15 by Peter-van-Tol

dixi83, 2023-04-21:
@Peter-van-Tol did you succeed in replacing the buffer IC's?

Peter-van-Tol, 2023-04-21:
Yes, I did moderate successfully replace the buffers. I screwed up the pads of one buffer beyond what could repaired with wires, so two Hub75 connectors don't work. But for my machine enough IO
[image]

If you have any troubles, please let me know for some advice.

dixi83, 2023-04-21:
Thanks for your reply!
1 question, did you also replaced them on the outputs or did left these original?
In my case I definitely did something wrong.. try to figure out what ;)
I received your stepgen PCB's! The parts are ordered, Thanks!

Peter-van-Tol, 2023-04-21:
I only replaced the buffers on the inputs. On a 5a-75e I wanted to have 8 connectors configured as output. So I replaced 6 buffers in total and left the output and the shared pins buffers untouched. See picture below for the modified buffers. 

[image]

For each buffer I've cut the trace for VCC and wired them to 3.3 Volts from the JTAG header (a Must, otherwise you feed 5 Volt to the Fpga, which releases the magic smoke). To configure them as output I also cut the trace to the direction pin and wired that to ground.

dixi83, 2023-04-21:
Thanks for the additional information, maybe I should have asked for this before frying my board...

Here is what I did: I replaced the output side (I marked them below, and the first one, u28 is already replaced here) as well... don't ask why ;-) I thought I read it somewhere.
[image]

This was my final (and fried) result:
[image]
I picked the 3.3V (for the VCC) from the JTAG and the GND (for the DIRection) from a nearby cap.

Peter-van-Tol, 2023-04-21:
Your solder skills rock! Way neater then what I ever produced and no ripped pads. 

Did you cut the traces (and check for connections) before soldering the buffers?

Next time, save the trouble of doing the output, that's not necessary.

If it is fried, I might have one in stock. Might arrive faster then ordering a new one from China.

dixi83, 2023-04-21:
Thanks for the compliment, In the past I followed some tutorial's on youtube. This one is one the better ones: https://youtu.be/naXglCTEtLw?t=123 This guy has more good video's. It's a mix of the right temperature (I rarely go above 300 degree), patience (very hard for me), enough flux and cleaning the old soldering pads with (In English I think it is) solder suck wire. 
But..... They might rock but they also fried my CPU :D
I really cut the old traces, checked and double checked them with my multi meter, there is no connection between the VCC pin and the 5v anymore... 
My 74LVC245's came from china... this could also be a cause. I ordered some new ones from Farnell, they are already here. 

My new board is is already in Liege (Luik) I expect it to arrive tomorrow or Saturday. Thanks for the offer!

Peter-van-Tol, 2023-05-04:
The maximum frequency will depend on the following factors:
- optocoupler. With 80 kHz they are relative descent. With my encoder I use differential lines, decoder and 6n137 optocoupler and this should be able to get to 1 MHz. There are even faster digital optocouplers which can support up to 10 MHz, but they are costly.
- firmware. When using this card with my  Litexcnc firmware, the clockspeed of the card is roughly 40 MHz. The pin is polled every cycle, so it will definitely be picked up with speeds up to 5 MHz. 

But how fast do you want your encoder to go? For my machine there is an encoder on the main spindle motor. Maximum speed is 3000 rpm and resolution is 2500 ppr. This gives a top frequency of 3000 x 2500 / 60 = 125 kHz....