q3k/chubby75 #109: What's 5A-75E V6.1?

ngmacha, 2023-02-09:
I've got an offer for the board marked as 5A-75E V6.1 - is it compatible and can be programmed with iCEStudio?

And, more general question, where can I find the difference between 6.0, 6.1, 7.1 and 8.0 boards? 

From the seller's picture seems this 6.1 board looks exactly as the 7.1
[image]

Any advise is very much appreciated

ngmacha, 2023-02-10:
Apparently, they sent the picture of the V6.1 card, but the actual card I've received is marked V7.1!!

Still, however, a table summarizing the differences between various V-s would be much helpful...

tomverbeure, 2023-02-10:
This GitHub repo is a great way for people to tell others what they’ve discovered, but not so great to ask for information because almost nobody is doing much with it.

If you discover anything new or different, don’t hesitate to share your finding by creating a pull request!