q3k/chubby75 #111: Blew up my first 5A-75E... what could have gone wrong...

dixi83, 2023-04-17:
Hi All,
My first attempt replacing the buffers with the 74LVC245 failed... after replacing them the CPu gets really hot and does not respond any more to anything.... so I guess I fried it.

[image]

I don't know what went wrong, I tested every IC after replacing for short circuit and between every pin.
There are 3 options I see:
[*] my multimeter fried something while testing
[*] the new 74LVC245 came from china and they are b/c/d (or maybe z) -stock
[*] The PCB is a 4 or more layer PCB and I cut to much

I ordered a new one but I don't want to fry my new one.

Thanks!

dixi83, 2023-04-19:
The version I have is a 8.2. As far as I could determine it has the same pin assignment as documented for the 7.1. I have only tested the RGB pins not A-E...

kittennbfive, 2023-05-08:
Hi,

at least on my board which is V7.0 some FPGA-pins are connected to multiple output buffers. This could lead to some short-circuits if not taken care of. Just a thought. Also maybe the buffers are crap, i would recommand buying them from a reputable source like Mouser, Digikey, Newark/Element14/Farnell, ... depending on your location in the world.

I have a file !!!FOR VERSION 7.0!!! which shows what is connected where. Il will add it to my github, might be useful for other people.

Also i would suggest removing all the buffers with hot air and see if the FPGA (not CPU) works again.

dixi83, 2023-05-09:
@kittennbfive thanks for your reply!
I ordered a new card from Ali and also some new IC's from Farnell. I will do a retry next week is I have more time. I will use kapton tape to isolate Pin 1 and 20 it looks really neat. I did not removed anything else then the buffers with a wire attached on the photo... what I do know now I should not have removed the buffers I wanted to keep as output. I did used a hot air gun. I only used a iron for cleaning the pads and some solder removal lint (not sure if it is the right English word)
Sorry for shouting "CPU" to a FPGA... I was not aware 🤣 

@everyoneelse this is the repo kittennbfive is talking about: https://github.com/kittennbfive/5A-75B-tools
I received my new card last week. I also ordered a mesa card to get started with my machine and getting used to the way of configuring LinuxCNC.

The thing I did not mentioned in my previous post is that I did not even test the card before changing anything. so another option would be that my first card was death on arrival...

kittennbfive, 2023-05-09:
The word you are looking for is "solder wick". ;-)

As i said, you really (imho) should remove (or erase at least) the FLASH containing the default bitstream so there won't be any FPGA-pins fighting against the new installed buffers. Also you should check for FPGA-pins connected to multiple buffers for the same reason. I am afraid i can't give more details because i don't have a V8.

dixi83, 2023-05-14:
Thanks for your comment and extra warning. I finally found what you meant: 
https://github.com/kittennbfive/5A-75B-tools/blob/8f4c14c88fb531b183f87d95d1defc01eb2f0712/I_want_inputs/README.md
The next question is a noob question ;-) ... 
Can I "remove" the flash by compiling "empty" firmware and load this using openFPGAloader or is there a way to do this in another easier way?

kittennbfive, 2023-05-25:
I would guess that simply erasing the flash should be ok, but honestly i don't know for sure. :(

closed 2023-08-29 by dixi83