q3k/chubby75 #117: Uploading bitstreams 5A-75B V8.0 SPI Flash

tvlad1234, 2024-01-20:
I just got my 5A-75B V8.2 and successfully swapped the output drivers on the right side of the board. I am able to generate bitstreams from VHDL designs and configure the FPGA over JTAG, with a CMSIS-DAP probe. My question is, is it possible to write custom bitstreams to the SPI flash and have it configure the FPGA on powerup?

tomverbeure, 2024-01-21:
Yes, it's easy to.

I have 2 rules in my Makefile: `upload` is to load the bitstream over JTAG, and `flash` is to program the bitstream into the spi flash.

```
upload: top.bit
	openfpgaloader  --vid 0x0403 --pid 0x6014 top.bit

flash: top.bit
	openfpgaloader  --vid 0x0403 --pid 0x6014 --unprotect-flash --write-flash top.bit
```

You'll obviously have to change the vid and pid to whichever device you're using.

tvlad1234, 2024-01-21:
I've just tried it, can confirm that it works!

closed 2024-01-21 by tomverbeure