q3k/chubby75 #123: Can Huidu HD-R712 be supported?

kholia, 2024-11-25:
Hey Folks,

This card is widely available for a reasonable price. I am wondering if this board has a JTAG header available, and if the onboard FPGA is well supported by FOSS toolchains.

Thanks for the help!

[image]
[image]
[image]
[image]

Sample product URL: https://ledcontrollercard.com/product/huidu-hd-r712-led-display-full-color-receiving-card/

PS: I haven't bought one yet.

tomverbeure, 2024-11-25:
The only way to know is to buy it and find out. It's not even clear what kind of FPGA is being used. At point, these receiving cards are so cheap that it may be better to just stick with what's already been done. Unless you want to do the reverse engineering for the sport, of course!

kholia, 2024-11-25:
Roger @tomverbeure - thank you!

The availability of these 'Huidu' cards seems to be better than the well-supported Colorlight boards - at least here locally.

If I fail to source more Colorlight boards, I might be tempted to get one of these Huidu boards and struggle to make it work.

Let's see!

kholia, 2024-11-26:
It seems Huidu R712 uses an [Anlogic](https://www.anlogic.com/en) FPGA and I failed at even downloading the required tool-chain as it is behind multiple layers of 'verification'.

kholia, 2024-11-30:
I was able to source Colorlight boards locally! Yep - Sticking with Colorlight makes sense for now. Thanks!

closed 2024-11-30 by kholia